springfall2008/predbat_addon #3: Cannot schedule new futures after shutdown

ivenhov, 2024-06-05:
I'm not sure if this should be here on in predbat repo

After successful migration from appdaemon version to addon everything was working as expected.
Full restart of the HA caused the following though

`
2024-06-06 00:37:06.423475: Info: Start socket for url http://supervisor/core/api/websocket
2024-06-06 00:37:06.424256: Error: Web Socket exception in startup: cannot schedule new futures after shutdown
2024-06-06 00:37:06.424599: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 15517, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 805, in _ws_connect
    resp = await self.request(
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 562, in _request
    conn = await self._connector.connect(
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 540, in connect
    proto = await self._create_connection(req, traces, timeout)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 901, in _create_connection
    _, proto = await self._create_direct_connection(req, traces, timeout)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 1155, in _create_direct_connection
    hosts = await asyncio.shield(host_resolved)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 874, in _resolve_host
    addrs = await self._resolver.resolve(host, port, family=self._family)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/resolver.py", line 33, in resolve
    infos = await self._loop.getaddrinfo(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 863, in getaddrinfo
    return await self.run_in_executor(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 821, in run_in_executor
    executor.submit(func, *args), loop=self)
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 167, in submit
    raise RuntimeError('cannot schedule new futures after shutdown')
RuntimeError: cannot schedule new futures after shutdown

2024-06-06 00:37:06.424742: Info: Start socket for url http://supervisor/core/api/websocket
2024-06-06 00:37:06.425475: Error: Web Socket exception in startup: cannot schedule new futures after shutdown
2024-06-06 00:37:06.425789: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 15517, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 805, in _ws_connect
    resp = await self.request(
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 562, in _request
    conn = await self._connector.connect(
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 540, in connect
    proto = await self._create_connection(req, traces, timeout)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 901, in _create_connection
    _, proto = await self._create_direct_connection(req, traces, timeout)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 1155, in _create_direct_connection
    hosts = await asyncio.shield(host_resolved)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 874, in _resolve_host
    addrs = await self._resolver.resolve(host, port, family=self._family)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/resolver.py", line 33, in resolve
    infos = await self._loop.getaddrinfo(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 863, in getaddrinfo
    return await self.run_in_executor(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 821, in run_in_executor
    executor.submit(func, *args), loop=self)
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 167, in submit
    raise RuntimeError('cannot schedule new futures after shutdown')
RuntimeError: cannot schedule new futures after shutdown
`
This was happening in a tight loop.
Restart of the addon fixed the problem.

springfall2008, 2024-06-09:
What version are you using, I thought I had fixed a few of those issues already?

ivenhov, 2024-06-09:
I was on 7.22.2 back then, currently on 7.22.5.
Just did the restart test and Predbat started OK so I guest that has been fixed.
I still see some warnings on startup but it settles down after 1-2 attempts and I know this is work in progress anyway.
I guess Predbat can ignore the errors if it knows HA's uptime is only few minutes? Just an idea.

Thanks for the great piece of software btw.

closed 2024-06-16 by springfall2008