springfall2008/predbat_addon #4: Predbat Docker

dandwhelan, 2024-06-09:
Hi, I've been trying to get Docker Predbat working on my work
[predbat.txt](https://github.com/user-attachments/files/15753875/predbat.txt)
[hass.txt](https://github.com/user-attachments/files/15753876/hass.txt)
 in progress docker raspberrypi. 

When I run the following... 

`docker run docker.io/library/predbat
`
I continuously get the below message. Included my predbat.py and hass.py in this issue.

Any help would be appreciated :) 

```
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'
Predbat crashed. Restarting in 5 seconds...
```

dandwhelan, 2024-06-09:
I feel like this might be bad timing for me to look at older documentation, as v8 has just been released. 

[image]

springfall2008, 2024-06-16:
I think you are missing the other files for Predbat v8?

dandwhelan, 2024-06-17:
Now v8 has been offically released I've tried again with all your updated files. Predbat starts, when I manually start the 
docker run docker.io/library/predbat I get this error. but everything else seems to work fine :) 

Starting Predbat standalone...
/config/predbat.py:10961: SyntaxWarning: invalid escape sequence '\s'
  res = re.search('THIS_VERSION\s+=\s+"([0-9.v]+)"', line)

springfall2008, 2024-07-06:
You can ignore that warning its fine.

BTW: Did you have to change anything in the Dockerfile.standalone?

dandwhelan, 2024-07-06:
Nah but I did have to update the time to BST London after an update. Also, after HA image updates, all the predbat switches reset to off.

dandwhelan, 2024-07-06:
Also on the instructions it said copy predbat.py. I know you've separated the files lately so I copied over all the .py files.

`Download predbat.py from Predbat repo (https://github.com/springfall2008/batpred/blob/main/apps/predbat/predbat.py) and place it into rootfs`

springfall2008, 2024-08-02:
Did this get resolved?

dandwhelan, 2024-08-02:
Yeah. I think I started again from scratch.

closed 2024-08-02 by dandwhelan