springfall2008/predbat_addon #5: Cannot install

ASTPlatinum, 2024-06-14:
I cannot install this addon. The following error is reported.

The command '/bin/bash -o pipefail -c apt-get install -y python3' returned a non-zero code: 100

HA Logs: Failed to to call /addons/6adb4f0d_predbat/install - The command '/bin/bash -o pipefail -c apt-get install -y python3' returned a non-zero code: 100

Can install other addons with no problems.

springfall2008, 2024-06-16:
I don't understand why Python3 isn't found, what system do you have?

ASTPlatinum, 2024-06-16:
> I don't understand why Python3 isn't found, what system do you have?

Running on docker.

If I jump into the container console of the supervisor you can see what version of Python is installed!

[image]

ASTPlatinum, 2024-06-20:
@springfall2008 Just forked, changed the following in dockerfile

RUN apt-get upgrade -y
RUN apt-get dist-upgrade -y

To

RUN apt-get update && apt-get install -y

And the install ran fine.

springfall2008, 2024-07-06:
I've made the suggested change in 1.1.8 but I don't understand why it would help

closed 2024-07-06 by springfall2008