springfall2008/predbat_addon #9: GivEnergy battery continues to charge when _battery_charge_rate set to zero when in ECO mode - setting behaviour change

DrJohnM61, 2024-07-06:
TLDR; setting the battery_charge_rate to zero to prevent charging requires Eco to be turned off to function correctly.

It would appear that setting the givtcp_..._battery_charge_rate to zero no longer stops the battery completely charging while in ECO mode as there is still a % of charge going to the battery. This seems to have been a change to the behaviour with a recent firmware update (although difficult to confirm).

When in Freeze discharging, the documentation states:
"Freeze discharging - The battery is in Discharge mode, the same as Idle (Eco) mode, but with charging disabled. The battery or solar covers the house load. As charging is disabled, if there is excess solar generated, the current SoC level will be held and the excess solar will be exported. If there is a shortfall of solar power to meet house load, the battery will discharge."

However, as noted above, there is still an amount of charging taking place. (Woodstock from the GivEnergy forum has also noticed this and reported: "On my system the battery consistently charges at a power level two higher than that set (unless the charging power is set to 0 when it charges three levels higher)". Note that I have reported this to GivEnergy through their "give feedback" system but without any response.

Turning Eco off from the GivEnergy cloud interface resolves the issue, however Predbat will override the setting back to Eco at some point.

Is there some way to configure Eco off if the charge limit is set to zero (or a non max value)?

springfall2008, 2024-07-06:
This is kind of expected, but Predbat supports pause mode which will stop the discharge/charge and works with 'new firmware' option in GivTCP if you want to try it.

BTW: I think you raised this ticket in the wrong project as it should be under batpred and not under the add on

DrJohnM61, 2024-07-07:
Thank you for your quick response and apologies for posting this in the wrong section.

It took me some time to find the pause_mode functionality that was missing in my configuration file. Now added this in and will monitor the change.

Thanks again.

closed 2024-08-02 by springfall2008