springfall2008/predbat_addon #10: Struggling to install

Boffinboy, 2024-07-07:
Hi There, I was switching to the add-on version from the original install. It wasn’t working properly, so I restarted home assistant, including add ons but I can no longer install v 1.1.8. I get the below errors
[image]
[image]

Boffinboy, 2024-07-07:
After further ingestions it seems the /addon_config/6adb45f0d_predbat only has a single file - hass.py in it - which I assume should have more in. Even if I try removing the repository and adding it back, I can’t seem to get it to install

springfall2008, 2024-08-02:
This is correct, once the build finishes it will launch startup.py in the container which downloads the remaining files. Did you try uninstalling and re-installing again?  When you uninstall, remove the 6adb4f0d_predbat directory and restart HA before attempting a new install and then make sure you wait long enough for the install to complete. After its clean then start the add-on for it to download the code.

Boffinboy, 2024-08-02:
> This is correct, once the build finishes it will launch startup.py in the container which downloads the remaining files. Did you try uninstalling and re-installing again? When you uninstall, remove [...]

Thank you. It took a few goes but managed to get it installed and working well. Apologies I should have closed the issue.

closed 2024-08-02 by Boffinboy