springfall2008/predbat_addon #12: Docker Documentation for Prebat_addon

nipar4, 2024-07-26:
Hi Trefor
  
  As discussed here is the documentation for the docker images on hub.docker.com

  They are all prebuilt images that will support both amd64 & arm processors (so Raspberry PI, most NAS & PCs).  They can be installed as either a fully enclosed container or a container with mounted external volumes and will populate the /config directory on creation and start-up.  To do this I have a run.standalone.sh start-up file which does the file copy on first boot - I do not use your current run.csh or run.sh start-up files although they are pretty much the same as I wanted to standardise on one start-up file that supported all architectures.

As mentioned the only issue I have is that with the template apps.yaml the logs throw lots of errors until a modified config file is added to the volume

Cheers
Nic

[Docker Instructions.docx](https://github.com/user-attachments/files/16394118/Docker.Instructions.docx)

nipar4, 2024-07-29:
Hi Trefor

  would you let me know what start.py does in the start routine?  I'm not really a programmer so am probably missing something but it looks to me like its downloading the program files from github which is also being done by hass.py, or at least it does for the standalone addon unless the routines are specifically for the HA addon?

Also if you use the attached Dockerfile it more the halves the size of the Docker image, dont see why it cannot be used for Home Assistant as well

Regards
Nic
[Dockerfile.standalone.txt](https://github.com/user-attachments/files/16414580/Dockerfile.standalone.txt)

springfall2008, 2024-08-02:
thanks, I've updated the documentation to include this: https://github.com/springfall2008/predbat_addon/blob/main/predbat/DOCS.md

springfall2008, 2024-08-02:
Startup.py literally downloads Predbat from the Github repository if it hasn't already been downloaded, the add-on itself does not contain the Predbat source code. Once it is downloaded it calls python to start Predbat

hass.py is the code that emulates some of the appdeamon functions that Predbat needs, basically it schedules the timer functions and restarts predbat if it crashes.

springfall2008, 2024-08-02:
One difference between standalone and the HA Add-on is that the add-on uses ubunto-base which is a build for Home Assistant, without that you won't get the SUPERVISOR key passed in and suchlike.

However I suspect I could remove python-dev and save some space as its not used.

nipar4, 2024-08-02:
So, in the Docker images should I use Startup.py or Hass.py as the entry point?

I am currently using hass.py as that’s what was in the Docker.Standalone dockerfile and that downloads the predbat source files as well so there might be a bit of duplication

I would rather make sure I use something your happy with so if anything changes its automatically reflected in the images

springfall2008, 2024-08-02:
The standalone dockerfile used run.csh which calls hass.py, but that does not download Predbat. Instead I was depending on the user downloading the files themselves and putting them into the install area manually. I think you should use run.sh which calls startup.py and downloads the code for your option 2. but for option 1 I don't think that works as apps.yaml will get overwritten each time?

nipar4, 2024-08-02:
I am doing the main file copying for the user as part of the image build except I am not currently running startup.py.  So does predbat.py do a file download of missing files as well? see the attached logs I am definitely getting the missing files downloaded and apart from the initial startup I am not getting any errors in the logs during normal operation

I'll test out startup.py

[_predbat-test_logs (4).txt](https://github.com/user-attachments/files/16471018/_predbat-test_logs.4.txt)

nipar4, 2024-08-02:
Do you expect all .py files to be in the /config directory?

springfall2008, 2024-08-03:
Predbat.py will download missing files but only if download.py is also present, its a safety mechanism if a user deletes a file by accident or a release download goes wrong.

springfall2008, 2024-08-03:
> Do you expect all .py files to be in the /config directory?

Yes

nipar4, 2024-08-05:
I have been testing the startup.py file as part of the docker image plus trying to make the installation as straightforward as possible for the user and have now got everything working in my environment.

To do this I've been looking at the predbat startup.py files and it looks to me  like it is being run from the root of the container then hass.py is being copied to the config directory and then run from there, I am guessing this is  because the working directory is being defined as config so python is looking there for hass.py and then downloading all python files to that location.

What I am proposing to do is copy the rootfs source files into a /addon directory then run startup.py and hass.py from there when the container is started.  This way all the python files are hidden from the user and don’t clutter up the /config directory

To do this  and allow me to use the startup.py we need to make a fairly small addition to the code so we can search in both the /addon directory and the /config directory and run hass.py from either context ensuring the file will work for both the Docker and the HA addon - I've also added the full file here for context……alternatively why not just do this for both the HA addon and Docker?
Is there any reason why the user would want easy access to the python files?

I have tested this for the last few days and it works in Docker but I do not know much about the HA addon structure

[startup.py.txt](https://github.com/user-attachments/files/16495834/startup.py.txt)

\# Define the specific directory for hass.py
hass_directory = "/addon"
hass_path = os.path.join(hass_directory, "hass.py")

\# Check if hass.py exists in the specific directory, if not, check the root directory
if os.path.exists(hass_path):
    print("Found hass.py in /addon")
elif os.path.exists(os.path.join(root, "hass.py")):
    hass_path = os.path.join(root, "hass.py")
    print(f"Found hass.py in {root}")
else:
    print(f"Error: hass.py not found in the directory {hass_directory} or {root}")
    sys.exit(1)

\# Run hass.py
print("Startup")
os.system(f"python3 {hass_path}")

So now I have two options on how to implement this when I build all the images.

If we make the above changes then I can leave all files in the /addon directory
This will be cleaner and remove the python files form the config directory
Plus I think this could also be done for the HA addon meaning both setups are the same

Or

As part of the container startup copy all python files into the config directory as this is where they all go now

If you want to see what I have done let me know and I'll upload a test image to the docker hub for you to look at

What do you reckon?

reidjr2, 2024-08-17:
EDIT: TL;DR.
The update of predbat via HA gui appears to fail when using the prebuilt images. It however works after a container restart.

Also took me a while to realise that I had to RTFM, and DELETE `template: True`. Commenting it out works in the original, so copying over apps.yaml wasnt working for me :-)
------------------------------------------

Your current approach seems to hard code the predbat version in to the predbat_addon docker image (?), breaking the HA predbat update. 

With the current pre-built images ( thank you !) I can attempt to use the "Predbat Core Update" from the HA gui when a new version is available. However the predbat version does not change, despite a log entry that it is complete. So to update predbat you need to update the docker container ?
With the original "build your own" image, and remapping /config for persistent storage, the updates work as expected, changing the predbat.py in the /config mapping.

```2024-08-17 10:56:53.684955: Predbat /addon/predbat.py version v8.3.1 currently running, latest version is v8.3.5 latest beta v8.3.5
2024-08-17 10:56:53.685004: Autoupdate: There is an update pending v8.3.5 Fix Broken core API when using HA URL - auto update is off
2024-08-17 10:57:38.851519: Requested install of latest version v8.3.5
2024-08-17 10:57:38.851899: Update Predbat to version v8.3.5
2024-08-17 10:57:41.547682: Update to version v8.3.5 completed
........
2024-08-17 11:00:00.505585: Predbat /addon/predbat.py version v8.3.1 currently running, latest version is v8.3.5 latest beta v8.3.5`

nipar4, 2024-08-17:
I take it your either using the alpine or the slim tags as I don't mess with the latest tag - Its not really hardcoded, I just rebuilt the images last night to include a more up-to date predbat version as the original builds were 8.3.1 I take the point though about it not restarting cleanly and needing a reboot, I didn't test that as that's built into predbat and not the docker image 
@springfall2008 I just tested on the latest image and that does not kill and restart predbat either, take it nothings changed for the restart on upgrade?

reidjr2, 2024-08-17:
I am using :Latest. I had wrongly assumed that it wasnt updating at all due to the log/no change in predbat. I only noticed after I had written the hard coded statement, and I added the EDIT comment.

Thanks for taking the time to build the images, and of course Trefor for all his hard work.

nipar4, 2024-08-17:
yea, my bad I updated all images and not just the alpine & slim images.  Should have left the latest tag alone.  I'll reupload that when I get back

nipar4, 2024-08-17:
BTW this will put it back to not looking for the template: True and therefore will run whatever, until I can work out how to automate the install properly

nipar4, 2024-08-17:
I'm doing this in stages, I have just uploaded the basic Dockerfile build so the latest tag is all back to the basic install @reidjr2 be nice if you can confirm it works the way your used to and you can upgrade/downgrade predbat.  Then I can add back in my changes to see what breaks the process (I do have a suspicion though)
I wont be doing any more changes until after the weekend

kneival, 2024-08-18:
Mine is not working at all after updating.
run.csh: No such file or directory. run.csh: No such file or directory.
Any ideas? Have tried a reinstall...

nipar4, 2024-08-18:
IF you are mapping a volume under config i.e. ~/config:/config as part of your compose or run command, you need to manually populate the folder from the rootfs directory on GitHub, once I redo the run script it will be automatic

kneival, 2024-08-19:
> IF you are mapping a volume under config i.e. ~/config:/config as part of your compose or run command, you need to manually populate the folder from the rootfs directory on GitHub, once I redo the run script it will be automatic

Thanks. I put all the contents of rootfs in to my config directory. It now starts but says :-

Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
ModuleNotFoundError: No module named 'predbat'

Maybe I am downloading the rootfs files from the wrong place?. There seems to be a lot more files in the config dir when its not mapped compared to when it is mapped.

I think I should wait until you have modified the run script?

nipar4, 2024-08-19:
OK I have uploaded a new version which contains the latest predbat version and will populate the config directory for you, be sure to pull from Docker hub to get the latest image.
As a precaution please delete the python files you copied over
Let me know how it goes

kneival, 2024-08-19:
All working thanks a lot. It copied all the files over and started!

files are missing, copying program files
'/addon/download.py' -> '/config/download.py'
'/addon/hass.py' -> '/config/hass.py'
'/addon/predbat.py' -> '/config/predbat.py'
'/addon/startup.py' -> '/config/startup.py'
modified apps.yaml present
Starting Predbat standalone...
starting startup.py...........
Update complete
Error: File /config/config.py is missing
Warn: Predbat files are not installed correctly, trying to download them
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.5/apps/predbat/predbat.py
Got data, writing to /config/predbat.py.v8.3.5
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.5/apps/predbat/config.py
Got data, writing to /config/config.py.v8.3.5
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.5/apps/predbat/prediction.py
Got data, writing to /config/prediction.py.v8.3.5
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.5/apps/predbat/utils.py
Got data, writing to /config/utils.py.v8.3.5
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.5/apps/predbat/inverter.py
Got data, writing to /config/inverter.py.v8.3.5
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.5/apps/predbat/ha.py
Got data, writing to /config/ha.py.v8.3.5
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.5/apps/predbat/download.py
Got data, writing to /config/download.py.v8.3.5
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.5/apps/predbat/unit_test.py
Got data, writing to /config/unit_test.py.v8.3.5
Downloaded files, moving into place
Bootstrap Predbat
Startup
Shutdown, sleeping 30 seconds before restarting
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
starting startup.py...........

reidjr2, 2024-08-26:
Just back from holiday. Looks like re-pulling latest ( which gave me 8.3.5) and then doing a predbat update through the HA GUI to 8.3.7 works as expected. Cool. I see the *.py files now appear in the mapped config director.
Good Job.

closed 2024-09-02 by nipar4