springfall2008/predbat_addon #13: predbat issue with config?

nblount, 2024-07-28:
After a reboot I am getting the following errors so predbat will not start, where do I need to look. I tried redownloading predbat no change.

  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10340, in load_current_config
    settings = json.load(file)
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 293, in load
    return loads(fp.read(),
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

2024-07-28 10:57:44.937859 WARNING pred_bat: ------------------------------------------------------------

nblount, 2024-07-28:
Did some digging and it appears that predbat_config.json was a zero length file, I have deleted it and restarted and we seem to be working again. Maybe a check in predbat.py for empty file?

springfall2008, 2024-08-02:
This isn't related to the add on as such but Predbat so will close it here.

closed 2024-08-02 by springfall2008