springfall2008/predbat_addon #14: Incorrect Import p(w/loss) calculation?

IWR69, 2024-08-27:
I've been using PredBat for just a few days so please forgive me if I'm being stupid but I believe the Import p(w/loss) column, as shown on the debug plan, is not being calculated correctly.

The value appears to be calculating incorrectly when setting either the Inverter Loss or Charge Loss settings.

To test I set Inverter Loss, Discharge Loss and Metric Battery Cycle Cost all to 0 and then set Charge Loss to 0.1 (10%).
13.85 + 10% is 15.24 but this is the result shown in the plan...
[image]

You see the same results if you set Inverter loss to 0.1 and Charge Loss to 0.

And this is the results when you set both Inverter Loss and Charge Loss to 0.1...
[image]

IWR69, 2024-09-10:
Thanks helps a lot

springfall2008, 2025-02-09:
Closing this issue as stale, please raise a new issue in the Predbat repository if you still have a problem

closed 2025-02-09 by springfall2008