springfall2008/predbat_addon #15: Simplified docker installation

nipar4, 2024-09-02:
@springfall2008 I have modified the Docker.standalone and made some changes to allow for a more simplified Docker installation as follows

Docker.standalone file modified to remove multiple RUN & COPY commands to reduce image size
removed Apt-get upgrade as the image should be maintained by the image owner unless upgrades are required by predbat - which they are not but can be added back in if needed
Removed unused ARGs and labels 
Added direct copy of apps.yaml & predbat files from the predbat git repository into build to ensure every build produces the latest predbat version and apps.yaml

Process to install is now very straight forward

1. git clone https://github.com/springfall2008/batpred.git
2. Change Directory to the predbat repository cd ./predbat_addon/predbat
3. Edit docker-compose.yaml to change the volume paths to match user environment
4. docker compose up -d
5. edit apps.yaml or copy modified apps.yaml to the config location

But I am not a git expert so how do I commit these changes for you to review?

Cheers
Nic

PianSom, 2024-09-05:
@nipar4 - apologies for the wildly off-topic post. Can't seem to find a way to DM you on GitHub ...

Have you used or played with [predai](https://github.com/springfall2008/predai) by any chance? If so, is there any chance you might have a pre-built docker container (and, ideally, its docker-compose) that you could share?

nipar4, 2024-09-05:
I've had a bit of a go and it produces something in the order of a 500gb image, however from what I can see it's only setup as an ha addon so could not get it to work

@springfall2008 is this possible for predai?