springfall2008/predbat_addon #16: Predbat crashed. Restarting in 5 seconds...

Snid0, 2024-09-04:
Hello,

I am trying to get the the prebuild docker running and having some troubles.

I have tried setting up the apps.yaml, but as soon as I remove the template line. the docker just seems to go into a loop where it crashes..

#################################################
Please Update apps.yaml
Once updated please delete line 13 'template: True'
#################################################
modified apps.yaml present
Starting Predbat standalone...
starting startup.py...........
Bootstrap Predbat
Startup
Shutdown, sleeping 30 seconds before restarting
Predbat crashed. Restarting in 5 seconds...

It doesnt seem to produce any logs or other error messages to indiacte what the error is or where to look.. I am not really sure what the next steps would be to try and resove this problem.

Thanks !

nipar4, 2024-09-04:
Need a bit more information: - is this a new installation?

- What docker tag are you pulling?
- There should be a bunch of files copied before you get to the update apps.yaml section
- What docker compose or docker run command are you using

Cheers

Snid0, 2024-09-04:
Hiya,

im using the following command to with the latest docker tag

docker run -d --name predbat -v /share/docker/appdata/predbat/config:/config:rw -v /etc/localtime:/etc/localtime:ro nipar44/predbat_addon:latest

Various files do get created in the config directory. This is where I am trying to configure the apps.yaml (using the solaredge template)
[/share/docker/appdata/predbat/config] # ls -ltr
total 904
-rw-r--r-- 1 admin administrators 114564 2024-09-02 17:25 inverter.py
-rwxr-xr-x 1 admin administrators   5339 2024-09-02 17:25 hass.py*
-rw-r--r-- 1 admin administrators  13063 2024-09-02 17:25 ha.py
-rw-r--r-- 1 admin administrators   4159 2024-09-02 17:25 download.py
-rw-r--r-- 1 admin administrators  37258 2024-09-02 17:25 config.py
-rw-r--r-- 1 admin administrators  14075 2024-09-02 17:25 web.py
-rw-r--r-- 1 admin administrators   6259 2024-09-02 17:25 utils.py
-rw-r--r-- 1 admin administrators  70294 2024-09-02 17:25 unit_test.py
-rw-r--r-- 1 admin administrators   2500 2024-09-02 17:25 startup.py
-rw-r--r-- 1 admin administrators  46040 2024-09-02 17:25 prediction.py
-rw-r--r-- 1 admin administrators 521348 2024-09-02 17:25 predbat.py
-rw-r--r-- 1 admin administrators  20866 2024-09-02 17:32 apps.yaml.orig
-rw-r--r-- 1 admin administrators  13850 2024-09-02 17:38 solaredge.yaml
-rw-r--r-- 1 admin administrators  14020 2024-09-04 09:49 apps.yaml

No log files seem to get created here though, all I get is the messages from the docker log which dont seem to be that helpful.

nipar4, 2024-09-04:
Looks like Docker is good to go, so must be something in your apps.yaml
Can you copy the original back and remove the template: true line and see if it gets any further, you should get a bunch of errors It wont work but it will prove if its your apps.yaml that is causing the problem

nipar4, 2024-09-04:
the only different between your setup and mine is the solaredge.yaml - is this a modified file?
If so remove it, or try removing it anyway
What platform are you on, not that it should be a problem as the startup script is handing over to startup.py and its crashing from there

Snid0, 2024-09-04:
the solaredge.yaml is just the orignal template file I downloaded. Ill try removing it but dont imagine it would make much difference.

Its running on a QNAP NAS. 

Its quite frustrating it doesnt seem to give any information as to what might be wrong. Ill try clearing out the config directory completely and starting fresh.

Snid0, 2024-09-04:
Hmm even after clearing the directory completely it is the same

starting startup.py...........
/config directory is empty, copying files
'/addon/config.py' -> '/config/config.py'
'/addon/download.py' -> '/config/download.py'
'/addon/ha.py' -> '/config/ha.py'
'/addon/hass.py' -> '/config/hass.py'
'/addon/inverter.py' -> '/config/inverter.py'
'/addon/predbat.py' -> '/config/predbat.py'
'/addon/prediction.py' -> '/config/prediction.py'
'/addon/startup.py' -> '/config/startup.py'
'/addon/unit_test.py' -> '/config/unit_test.py'
'/addon/utils.py' -> '/config/utils.py'
'/addon/web.py' -> '/config/web.py'
'/addon/apps.yaml' -> '/config/apps.yaml'
#################################################
Please Update apps.yaml
Once updated please delete line 13 'template: True'
#################################################
#################################################
Please Update apps.yaml
Once updated please delete line 13 'template: True'
#################################################
modified apps.yaml present
Starting Predbat standalone...
starting startup.py...........
Bootstrap Predbat
Startup
Shutdown, sleeping 30 seconds before restarting
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
starting startup.py...........

nipar4, 2024-09-04:
> I don't think you should have solaredge.yaml and apps.yaml in the same directory, I think they will both get bound by Python to predbat.py (as they have the same module: predbat in them) and this will cause confusion.
> 
> If you have a solaredge configuration then and want to use that as your predbat config, then you need to copy solaredge.yaml over the top of apps.yaml. The template apps.yaml supplied with Predbat is for GivEnergy inverters. If you have a different inverter type then you replace the template with the right XXX.yaml for your setup - see https://springfall2008.github.io/batpred/other-inverters/

It has to be a conflict somewhere as Docker is configured OK

gcoan, 2024-09-04:
Definitely no predbat log file written to /config or where you are running docker from?  

Can you set docker into debug mode to trace what it's actually doing?

Snid0, 2024-09-04:
No its not creating any logfile from what I can tell. Just the python and yaml files that get downloaded initially. 

Not sure what I need to do to put it into debug mode? Is there a setting I need to enable?

nipar4, 2024-09-04:
I think the Docker boot is complete and it has handed off to python and startup.py so the problem appears to be in predbat
I'm inclined to go with what was said above and its a conflict between apps.yaml and solaredge.yaml.  I just looked on gthub and solaredge.yaml is basically apps.yaml by another name
So remove solaredge.yaml first and see what happens

Snid0, 2024-09-04:
I cleared the config directory completely (including the solaredge.yaml) and let it redownload all the files. Unfortunatley it didnt make any difference :( Still seeing exactly the same behaviour.

nipar4, 2024-09-04:
What model is the qnap?

Just for kicks try the nipar44/predbat_addon:slim-latest

Snid0, 2024-09-04:
It's a TS-673

I will give that a go!

Snid0, 2024-09-04:
I did try the slim version and it didnt make any difference :(

From what I can tell it gets as far as running hass.py which should create the predbat.log but obviously it errors out before it manages it.

Snid0, 2024-09-04:
I have done a little messing around by changing the end of statup.py from:

print("Startup")
os.system("cd " + root + "; python3 hass.py")

to:

print("Startup")
status = os.system("cd " + root + "; python3 hass.py")
print(status)

Now when start the docker it seems the return code from this command is 32512

However if I open up a shell for the docker and run the following command..
cd /config; python3 hass.py

It seems to work (albeit with lots of errors)

So the issue seems to be with the startup.py script?

nipar4, 2024-09-04:
You will get lots of errors because your apps.yaml is not configured, but at least that confirms the docker is OK and the container sees the files in /config
@gcoan doubt it but could this be in part to do with the code to try and find addon config folder?

Doesn't qnap have its own container management app?

gcoan, 2024-09-04:
> @gcoan doubt it but could this be in part to do with the code to try and find addon config folder?

I doubt it too because that code is later in the ha.py startup. It makes an API call to HA supervisor to get the add on slug id and if that fails then it just leaves the path set to the default /config. And you'd have seen a bunch of logging by then. 

Was wondering what the config of the ha_url is in apps.yaml, whether this isn't pointing correctly to the HA IP address, but again I'd expect more verbose output? But check this. 

Maybe need to trace what happens in the startup script, something fatal I expect around the initial startup such as no permission to write the log file🤷‍♂️

Snid0, 2024-09-04:
it seems to fail on..

os.system("cd " + root + "; python3 hass.py")

the return code from this command is 32512

nipar4, 2024-09-04:
Could be permissions to the /config directory, that would affect both builds

Snid0, 2024-09-04:
I think it may still be an issue with the docker / pythan. Im no expert so dont really know what the issue is. Hoever if I open a shell into the predbat docker and then launch python and run the os.system command I get an error.. It doesnt matter what command I try to run with os.system I always get the same error.

# sudo docker exec -it predbat sh
# python3
Python 3.11.4 (main, Dec  7 2023, 15:43:41) [GCC 12.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.system("ls")
32512

If i run the same steps on my docker which runs homeassistant the ls commands works fine.

gcoan, 2024-09-04:
searching for that error code from python

https://stackoverflow.com/questions/48484152/os-system-returns-error-code-32512-python

suggests one source of the error code is that the PATH for the os command isn't set

try
os.system("/bin/ls")    [or wherever the commands are installed]

to disprove this guess

Snid0, 2024-09-04:
Nope that doesnt seem to help..

```
# which ls
/usr/bin/ls
# python3
Python 3.11.4 (main, Dec  7 2023, 15:43:41) [GCC 12.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.system("/usr/bin/ls")
32512
>>> os.system("/bin/ls");
32512

```

nipar4, 2024-09-05:
Try running the container without mapping the config directory
docker run -d --name  -v /etc/localtime:/etc/localtime:ro nipar44/predbat_addon:latest

Snid0, 2024-09-05:
Hi,

I have tried removing the mapping directory, unfortunately it doesnt seem to make a difference. :(

nipar4, 2024-09-05:
Same error?

If so then does qnap do any resource management?

The image happily runs standalone on amd and arm systems, but I don't know what impact qnap has

Snid0, 2024-09-05:
OK I seem to have got it working. I updated "container station" QNAPS wrapper package for docker. And its now working (with lots of errors)!

Now I just have to try and sort all these errors :)

Thanks for your help and suggestions!

nipar4, 2024-09-06:
Sounds like the main issue is qnap docker management
The other errors will disappear when the apps.yaml is configured

springfall2008, 2025-02-09:
Closing this issue as stale, please raise a new issue in the Predbat repository if you still have a problem

closed 2025-02-09 by springfall2008