springfall2008/predbat_addon #17: Documentation update for migrating from the old HA AppDaemon installation to the new add-on

iainfogg, 2024-09-07:
I couldn't find anything about moving from the AppDaemon add-on to this one in the documentation page - may be worth adding it as I expect there will be others too trying to make the move at some point.

I know I've seen the info on a Facebook post so I'll keep hunting for now, but getting the doc updated would be good too.

Looking forward to trying it out!

gcoan, 2024-09-07:
> I couldn't find anything about moving from the AppDaemon add-on to this one in the documentation page - may be worth adding it as I expect there will be others too trying to make the move at some [...]

It's already there in the installation doc, right at the end 

https://springfall2008.github.io/batpred/install/#upgrading-from-appdaemon-to-predbat-add-on

iainfogg, 2024-09-08:
Thanks @gcoan for pointing me to that, very useful!

I was actually meaning the add-on documentation page doesn't mention migration as an option for installing the add-on, I guess the add-on doc page just needs a mention and that link you posted.

gcoan, 2024-09-08:
Added to Predbat add-on documentation as per suggestion.

PR #19

iainfogg, 2024-09-09:
Great, thanks a lot @gcoan

closed 2025-02-09 by springfall2008