springfall2008/predbat_addon #20: 2x Errors: "Exception raised 0 preventing" which seems to prevent PredBat running

silkit, 2024-09-19:
Hi,

I am trying to get the add-on working with a Fox Inverter. The yaml template for the standalone Predbat doesn't work, but i have created my own which seems to be working with the add-on.
(i will post up once i get the last couple of errors resolved).
Nearly all the sensors seem to be working (or were initially working). Currently running in Monitor and Read-Only. However, the plan seems to be not updating. It only appears to have updated last night when i got all the sensors initially working (or so i thought!)
[image]
[image]
[image]

But not updating since yesterday:
[image]

Looking at the add-on logs. i think it seems to be caused by the two below errors.

The first one seems to relate to the Carbon Intensity sensor. I have the default set in the yaml of "sensor.carbon_intensity_uk" which is working in HA with the Carbon Intensity Add-On already installed, but i don't see an associated sensor "predbat.carbon_now" which is showing as the error, has been created. The only carbon related sensor i can see currently that predbat has created is the "predbat.carbon_now".
The second error - I'm afraid i have now idea about :-(

`2024-09-19 10:50:07.809656: Warn: No carbon intensity history in sensor predbat.carbon_now
2024-09-19 10:50:37.358265: Error: Exception raised 0
2024-09-19 10:50:37.358879: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 11116, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 9889, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9403, in fetch_inverter_data
    curve = inverter.find_charge_curve(discharge=False)
  File "/config/inverter.py", line 459, in find_charge_curve
    charge_rate_data[0],
KeyError: 0

2024-09-19 10:50:37.422375: Info: record_status Error: Exception raised 0
2024-09-19 10:50:37.422593: Error: 0
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 11121, in run_time_loop
    raise e
  File "/config/predbat.py", line 11116, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 9889, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9403, in fetch_inverter_data
    curve = inverter.find_charge_curve(discharge=False)
  File "/config/inverter.py", line 459, in find_charge_curve
    charge_rate_data[0],
KeyError: 0`

Currently running
HA Core 2024.9.2
Supervisor 2024.09.1
Operating System 13.1
Predbat 8.4.8

Please be gentle with me lol. I'm not very technical but think i get the basics!
Any suggestions to resolve at all please?
Thanks

gcoan, 2024-09-19:
In the documentation it says:

input_number.predbat_carbon_metric (carbon enable) When Carbon footprint tracking is enabled (switch.predbat_carbon_enable) you can specify a cost per Kg of CO2 used to weight the selection of plans. Values of around 10-200 will give varying outcomes to trade off cost vs carbon footprint of your system.

Note: Carbon footprint tracking can only be enabled if apps.yaml is configured to point to the correct CO2 cost sensor

Have you got apps.yaml correctly configured to the CO2 cost sensor?  Maybe for now, turn switch.predbat_carbon_enable off and see if Predbat runs OK, we can then debug the carbon stuff once the core Predbat is running

silkit, 2024-09-19:
Hi @gcoan 

In the Predbat yaml i have have used the sensor as per the guide:
[image]

This is what the sensor is showing:
[image]
[image]

I have now switched off the "carbon enable" switch and rebooted HA and subsequently rebooted the add-on again separately. (I agree it makes sense to disable that at the moment)

I have captured the full boot log here:

`Bootstrap Predbat
Startup
 Add-on: predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.2.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.1  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.9.2
 Home Assistant Supervisor: 2024.09.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
Predbat files are installed correctly for version v8.4.8
[15:18:25] INFO: Configuring timezone (Europe/London)...
**** Starting Standalone Predbat ****
2024-09-19 15:18:25.431954: Loading apps.yaml
2024-09-19 15:18:25.842063: Info: Connected to Home Assistant at http://supervisor/core
2024-09-19 15:18:25.857818: Info: Add-on slug is 6adb4f0d_predbat
2024-09-19 15:18:25.858143: Creating task: <coroutine object HAInterface.socketLoop at 0x7f888e31b0>
2024-09-19 15:18:25.860314: Info: Start socket for url http://supervisor/core/api/websocket
2024-09-19 15:18:25.866230: Info: Web Socket task started
2024-09-19 15:18:25.867903: Creating task: <coroutine object WebInterface.start at 0x7f888e03c0>
Web interface started
2024-09-19 15:18:25.879638: Warn: unable to find /addon_configs/6adb4f0d_predbat/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
s6-rc: info: service base-addon-log-level successfully started
2024-09-19 15:18:25.896350: Info: Web Socket active
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Watching ['./download.py', './apps.yaml', './hass.py', './utils.py', './config.py', './predbat.py', './prediction.py', './ha.py', './web.py', './inverter.py', './unit_test.py'] for changes
2024-09-19 15:19:09.279865: Error: Exception raised 0
2024-09-19 15:19:09.362481: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 11065, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9889, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9403, in fetch_inverter_data
    curve = inverter.find_charge_curve(discharge=False)
  File "/config/inverter.py", line 459, in find_charge_curve
    charge_rate_data[0],
KeyError: 0

2024-09-19 15:19:09.414124: Info: record_status Error: Exception raised 0
2024-09-19 15:19:09.414321: Error: 0
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 11071, in update_time_loop
    raise e
  File "/config/predbat.py", line 11065, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9889, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9403, in fetch_inverter_data
    curve = inverter.find_charge_curve(discharge=False)
  File "/config/inverter.py", line 459, in find_charge_curve
    charge_rate_data[0],
KeyError: 0
`

Predbat plan has now gone to Null:
[image]

Error exception still showing in line with the logs (Albeit i see for some reason it hasn't updated the error time/date stamp from the reboot as few minutes ago:
[image]

gcoan, 2024-09-19:
OK thanks, looks like the carbon intensity was just a warning and wasn't relevant.

The errors immediately after the carbon intensity warning in the original log and repeated in the current log suggest that Predbat is failing in trying to calculate the battery charge and discharge curves.  You probably don't have the sensors setup correctly with this data from your Fox inverter

In your apps.yaml do you have something like this:

```
  battery_charge_power_curve:
    auto

  battery_discharge_power_curve:
    auto
```

Suggest you comment all four of these lines out so it looks like this:

```
#battery_charge_power_curve:
#    auto

#  battery_discharge_power_curve:
#    auto
```

Then predbat will stop trying to create the charge/discharge curves.

silkit, 2024-09-19:
Interesting. I looked in the yaml and i had commented out the charge curves. But that got me looking at the other predbat log in the config file and i saw somewhere under the script that it was still trying to calculate an auto charge curve (which i think in the instructions it mentions that it may still try to do this if it is disabled in the yaml).
I noticed of the 3 or 4 sensors this was auto charge curve was using, one of them was a numerical one rather than a sensor. Checking back over my config i had used a fixed numerical value for the "charge_rate" sensor i.e. 5000 (to reflect 5Kw max charge) as with the Fox the charge rate sensor is in Kw and Predbat apparently needs Watts. This made me think again about using a fixed value.
I have subsequently created a template sensor helper to simply take the value of the existing sensor and then convert the value from Kw to W, and then used that new sensor in the yaml instead of the numeric value i had used.
Rebooted and it started without any errors and it created a plan. Status is showing no errors and i checked the log in the add-on and no errors showing.
I subsequently went back and switched on the Carbon Footprint and re-booted again - It started up fine and loaded the other sensors and no errors either!

Thank you for your help and support.
It looks like the inverter values have to use a sensor rather than a numerical value.

Time to leave it settle down and run for a couple of days now :-)

auto calculation charge_rate:

gcoan, 2024-09-19:
> I noticed of the 3 or 4 sensors this was auto charge curve was using, one of them was a numerical one rather than a sensor. Checking back over my config i had used a fixed numerical value for the "charge_rate" sensor i.e. 5000 (to reflect 5Kw max charge)

Great news you've got Predbat working.

Yes, what you did with the charge rate sensor would have caused Predbat problems.  What Predbat is doing in the charge/discharge curves is to step back through historical charging/discharging data and see how the battery actually charged/discharged regardless of the set charging rate.
To expand a bit further using my own battery as an example, when charging at max rate (2600W), as the state of charge gets towards 100% the battery slows down charging, for example charging at 2300W to 92% then 2000W to 94% then 1500W to 96% etc.   Its this "real world" charging profile that Predbat is seeking to discover so it can model it in the predbat plan - its only about getting the plan more accurate, Predbat will still charge/discharge at max rate.

So you setting it to a fixed value would cause it (a) to have no history and (b) not be reflective of how the battery actually charges.  By creating a template sensor you've got around the first issue, but the data supplied isn't going to help with part b.
TBH though not having an accurate charge/discharge curve isn't the end of the world, it just means that the plan model is a bit less accurate.

I will add something to the documentation to make it clearer how these sensors work and how they should be populated for non-GE inverters.

Cheers!

silkit, 2024-09-20:
Just on that point around using your own battery. It is still in theory using a fixed value. That existing sensor entity from Fox that was in Kw, that i created the conversion template to show W. This sensor is user adjustable by overwriting the Kw number, but is left at the maximum of 5Kw so i don't think it will still create a charge profile then on that basis, so i guess i will still have an issue with point (a) and (b) above. but as you say i can probably live with a slightly less accurate charge curve if it is the only thing that it affects.

Just a couple of follow up questions please.......

1) My batteries are capable of 90% depth of discharge. I have set the "battery_scaling" in the yaml to "0.9", but then i have also used the predbat input variables to put in a value in Kw for "Best SOC Keep" = 2.2kWh and "Best SOC Min" = 2.1kWh (The batteries are 20150wh so those numbers roughly translate for the 10% floor whilst having the "keep" slightly above the "min" as suggested in the documentation) - Should i be setting this in both the yaml and the predbat input variables? Do i need one or the other?

I am wondering could this be why i am seeing a slightly different SoC in Predbat "current state of charge" sensor as opposed to the battery SoC sensor?
This is slightly complicated by the fact that Fox inverter doesn't seem to expose a battery kWh remaining sensor - it only has the SoC % - So i had to create my own sensor for the KwH remaining to use in the predbat yaml for the "soc_kw:"
`{{ states('sensor.battery_soc') | float / 100 *
states('sensor.bms_kwh_remaining') | float }}`
<details><summary>My Created Sensor</summary>
<p>

[image]
</p>
</details> 

<details><summary>SoC Power Sensor in Predbat</summary>
<p>

[image]
</p>
</details> 

2) In the yaml i have used the services to control charging option:
# Services to control charging/discharging
  charge_start_service:
    service: select.select_option
    entity_id: "select.work_mode"
    option: "Force Charge"
  charge_stop_service:
    service: select.select_option
    entity_id: "select.work_mode"
    option: "Self Use"
  discharge_start_service:
    service: select.select_option
    entity_id: "select.work_mode"
    option: "Force Discharge"
now further down in the yaml there is the option of "scheduled_charge_enable" & "scheduled_discharge_enable"- do these entries need to be populated or can it be commented out. The reason i ask is that there is technically a switch in the Fox inverter entries with a toggle button on a dashboard to enable "charge from grid" with a schedule being set, but looking at the entity this is linked to, it is exposed as a binary_sensor so i don't think this would work if i used it would it?
(My lack of knowledge not understanding how a toggle switch can be represented by a binary_sensor i think)
The other issue with the Fox is that to enable a timed charge manually, there are in fact two toggle buttons that need to be turned on :
<details><summary>Fox Charge Period Card</summary>
<p>

[image]
</p>
</details> 
If only the "enable charge period" is toggled on - what this does is to freeze the battery at the current SoC and then draw from the grid. To ensure the battery is charged from the grid, you then have to also toggle on the "enable charge from grid".
Ideally i was thinking if i can do away with using these entries, would predbat control the timings of the charge etc based on it's own timings and control this by using/changing the work modes above?

gcoan, 2024-09-20:
> 1. My batteries are capable of 90% depth of discharge. I have set the "battery_scaling" in the yaml to "0.9", but then i have also used the predbat input variables to put in a value in Kw for "Best [...]

If your battery correctly reports its size as 20150Wh then you don't need to set battery_scaling to 0.9 as Predbat will then scale the reported battery size and that will throw the SoC figure off.   battery_scaling is only used when the inverter doesn't report the size correctly.

To model the 90% DoD, the most important thing is to set best_soc_min. I would configure it in both apps.yaml and the HA input control.  best_soc_keep is less important, its a 'soft' limit that predbat will try to keep in the battery (e.g. to cater for unexpected high load), so slightly above best_soc_min is fine.

> If only the "enable charge period" is toggled on - what this does is to freeze the battery at the current SoC and then draw from the grid. To ensure the battery is charged from the grid, you then [...]

I'm sorry I don't know the answer to this question.  Probably need some trial and error on your part or see what other Predbat users have done for their Fox inverters

silkit, 2024-09-25:
Thank you. I have removed the battery scaling from the yaml.

I did then set the "best_soc_min" in both the yaml and input control - This was until i noticed that in the configuration yaml there is a an entity i missed; "has_reserve_soc:" which i switched from False to True. This is linked to the "reserve" yaml input and the "'input_number.predbat_set_reserve_min" entity which i can then just point it to my inverter min SOC sensor.
<details><summary>PredBat Input Variable Exposed</summary>
<p>

[image]
[image]

</p>
</details> 
and seems to be being used for something as it's picking it up in the logs. However, the logs also now seem to be showing some conflict with the Battery_min previously set.

You can see it captures the 10%, but it also still captures the battery min SOC.
`2024-09-25 12:30:25.631890: Reserve min: 10.0% Battery_min:4.0%
`

I thought i would try and override that 4% to also be 10%, but i can't seem to find an entity exposed in HA any longer named "battery_min" anymore - i assume this was the "input_number.predbat_set_reserve_min" entity which is no longer in the PredBat Input Variables either as it seems have have been removed/replaced by the inverter min SOC one and i don't have the battery_min in my yaml now either.

Should Predbat be using multiple minimums in this way and which one takes priority?
I have tried to add "Battery_min: 10%" back into the yaml as a workaround to override it, but it hasn't worked. It remains at 4%.
Obviously it's not going to do the batteries much good going below the 10% threshold

But......

There also seems to be another issue. Because of enabling that inverter reserve SOC, this has now led on to another anomaly. In that i still have the "best SOC keep" &  "best SOC Min" variables in predbat that i originally keyed in the KwH numbers (to mimic the 10% min SOC) and these also seem to be being used as well as shown in the log?!
`2024-09-25 12:30:25.633982: Found 1 inverters totals: min reserve 2.02 current reserve 2.02 soc_max 20.16 soc 14.52 charge rate 5.0 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 5.0 export limit 5.0 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
`

In summary - I seem to now have 4 battery minimum limits set and according to the logs, are being acknowledged in some shape or form so i assume will be being used in the calculations in the background.

1. Inverter linked Minimum SoC = 10% - Working and changes if i change the minimum on my inverter.
2. Battery_min = 4.0% (which was originally keyed in both yaml and the input variables (removed from yaml and not in the input variables to change anymore and i can't seem to override by putting the entity and a new value in the yaml))
3. best SOC keep - exposed as manual kWh number
4. best SOC Min - exposed as a manual kWh number

Which of the 4 battery reserves is PredBat using and which takes priority here?
I would ideally like to just use the inverter min SOC as the % value to keep it simple, rather than have to mess around calculating the kWh value if i want to move from say 10% to 15%, and it allows for only one input variable, not several which it doesn't really need. But obviously i don't know what it is using in it's calculations anymore.

I have managed to sort the inverter control for Fox and predbat is now running in full control of charge/discharge for the last couple of days without issue and has been adhering to the plan, but my battery as yet hasn't been emptied and with only cheap rates overnight it is difficult to test during the day as i don't want to incur the cost to then recharge the battery for use the rest of the day.

Thanks

silkit, 2024-10-15:
so the battery minimum at 4% has now disappeared. I used the entity "battery_min_soc" in the YAML and pointed this to my inveter SoC.
The sensors from point 3 & 4 (best SOC keep - exposed as manual kWh number & best SOC Min - exposed as a manual kWh number) i have removed removed the values from the Predbat GUI input, not sure how they were being handled in the code, but using the "battery_min_soc" seems to be working.
I have also noticed that the "Set reserve enabled" switch in the advanced GUI reflects the Inveter min SoC. You change the value in the inverter and the value is updated in the Predbat GUI "Set Reserve Min" and also works when you change the value in the "Set Reserve Min" so it updates the inverter min SoC. I don't know if this is the case across all inverters but it's not clear in the documentation of the affect.
Whilst i don't know what affect the various/multiple reserves had on the calculations and which Predbat prioritizes, i have managed to remove the additional one's and it's working ok and honoring the min SoC.
I think this is resolved now to the extent it's working but the "unkown" is the affect and priority of all the other reserve options in Predbat.
Thank you again for your help on this :-)

closed 2024-10-15 by silkit