springfall2008/predbat_addon #21: Balancing 2 inverters / bateries

discodancerstu, 2024-09-19:
[predbat.log](https://github.com/user-attachments/files/17059331/predbat.log)

As mentioned, recent log whilst running GivTCP V3, inverter balancing code V6.

When disabling balancing V6 automation, batteries stopped balancing and diverged swiftly.

springfall2008, 2024-09-22:
There's something odd here, notice that inverter 0 is charging which inverter 1 is discharging, hence the cross discharge is reported:

2024-09-19 11:49:16.268814: BALANCE: socs [86, 73] reserves [0.4, 0.4] battery_powers [-935.0, 289.0] total -646.0 battery_max_rates [2600.0, 2600.0] charge_rates [2570.0, 2570.0] pv_power [2357.0, 2264.0] load_power [1281.0, 2458.0] total 5140.0 discharge_rates [2570.0, 50.0] total 2620.0

The discharge rate of inverter 1 is set to 0 but comes back as '50' which is preventing it from being effective.

What inverter do you have and any ideas why we are getting 50 and not 0?

discodancerstu, 2024-09-22:
Hi Trefor,

I'm running a dual inverter and battery system, 2x Gen 3 GE x 5kw inverters and 2 x 9.5 batteries.

I was running the previous version of GivTCP for months with zero issues, this has now cropped up when V3 was installed.

I've not changed any entities or settings anywhere.

springfall2008, 2024-09-22:
Can you see if it works on GivTCP vs 2.x as I suspect thjs is the issue?

discodancerstu, 2024-09-25:
Sorry Trefor, I deleted v 2.x and have not kept a backup.  I'm not confident reinstalling it as don't want to create any further issues.

However, before the V3 update, PB was working perfectly, no warnings or errors.  

I followed the V3 install guide and made no PB changes apart from the apps.yaml code to point to v3, and we are where we are now unfortunately.

springfall2008, 2025-02-09:
Closing this issue as stale, please raise a new issue in the Predbat repository if you still have a problem

closed 2025-02-09 by springfall2008