springfall2008/predbat_addon #23: Warning in Supervisor Logs

gt008, 2024-10-02:
Hi I am getting the following warnings in the HA supervisor logs. Predbat appears to be working but unsure of the significance of these log warnings. 

Logger: homeassistant.components.recorder.db_schema
Source: components/recorder/db_schema.py:625
integration: Recorder (documentation, issues)
First occurred: 11:33:59 (201 occurrences)
Last logged: 13:59:40

State attributes for predbat.pv_energy exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.load_power exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.load_power_best exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.best_load_energy exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.plan_html exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored

This maybe somehow related to PredAI as first started seeing this error when I installed PredAI. 

Thx. Giles

gt008, 2024-10-02:
Ok. Thx. I only have the forecast hours set to 24 which is less than the default but seems like it is nothing to worry about. Thank you. Giles

gcoan, 2024-10-02:
It is indeed nothing to worry about, it's just a warning and you can suppress it as per the FAQ's.

closed 2024-12-29 by gt008