springfall2008/predbat_addon #24: predbat_solis_solax_modbus_0_charge_limit

gt008, 2024-10-03:
Is this an issue? If not what is best delete or ignore? 

[image]

gt008, 2024-10-08:
Hi  This issue is listed as a “repair” in my settings in HA alongside any updates. It is therefore sticky. It is not simply an entry in my Supervisor logs. I can simply delete the history but it seems like a more important issue than a warning in one of the logs. Or is this just a quirk in Home Assistant?

gcoan, 2024-10-08:
It would indicate that whatever created the sensor (which based on the name looks like it was either predbat or it was the Solis modbus integration) originally created this with a state class and now it doesn't have a state class so you are getting this warning that HA can't continue accumulate long term statistics for this entity

To be honest though, charge limit, is probably not something you really need to keep long term statistics forever for so you can safely remove the history

gt008, 2024-12-29:
Resolved in recent update. Thx

closed 2024-12-29 by gt008