springfall2008/predbat_addon #25: Predbat takes 12-13 mins to update the plan and inverter state when moving to a new 30 min band

KaliPete, 2024-10-18:
Note that I have been seeing this issue for a few days - I did just run the latest Octo & HomeAssistant updates but there doesn't appear to be any impact from those.
I also ran a final test where I cleared all the manual charge/freeze slots I had configured before the next 30 min band - it looks like it still took about 12 mins to run, so maybe it just takes that long to run everytime any variable is changed? It did feel like that earlier today when I added some force idle and it seemed to take forever to apply - though I wasn't timing that.
[predbat.log](https://github.com/user-attachments/files/17440773/predbat.log)

KaliPete, 2024-10-19:
I found what I think might be a related error in the HA logs;

Log Details (WARNING)
Logger: homeassistant.components.recorder.db_schema
Source: components/recorder/db_schema.py:624
integration: Recorder ([documentation](https://www.home-assistant.io/integrations/recorder), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+recorder%22))
First occurred: 18 October 2024 at 21:28:08 (2431 occurrences)
Last logged: 21:57:14

State attributes for predbat.soc_kw_best exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.battery_cycle_best exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.load_power_best exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.best_load_energy exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.plan_html exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored

KaliPete, 2024-10-19:
[image]
[image]
Also, attached are 2 examples where it took approx 12 mins to update to the new scheduled state

KaliPete, 2024-10-19:
Ah, I can see in my last log that it's the recorder making the log so big - SO!
Testing: Recorder reduced to 15 days. I will share the impact results.

> I found what I think might be a related error in the HA logs;
>
> Log Details (WARNING) Logger: homeassistant.components.recorder.db_schema Source: components/recorder/db_schema.py:624 integration: [...]

KaliPete, 2024-10-19:
After updating the recorder to 15 days, I'm down to 90 seconds to change state now - so most likely was just this.
Happy to close whenever you are.

gcoan, 2024-10-19:
> I found what I think might be a related error in the HA logs;
>
> Log Details (WARNING)
> Logger: homeassistant.components.recorder.db_schema
> Source: components/recorder/db_schema.py:624
> integration: [...]

this is not the cause of your problem, this is a 'known limitation' of Predbat within HA and you can suppress these messages

See the FAQ's https://springfall2008.github.io/batpred/faq/#predbat-is-causing-warning-messages-about-exceed-maximum-size-in-the-home-assistant-core-log

But per your later comments, if you do store too much history data in the HA database it will affect things.  Try to keep the database down to a manageable size

closed 2024-11-02 by KaliPete