springfall2008/predbat_addon #27: SolarEdge - Mode resets after command timeout time

leacho73, 2024-11-06:
Since Sunday - 06:30 - I've noticed that my SolarEdge battery will return to its default mode 1 hour after the last command that predbat sent to it - for example - if I plug the car in - predbat will start to charge the battery - but 3600 seconds later the battery will return to maximise self consumption - not sure if this is a predbat or SolarEdge integration issue - but never had this problem until Sunday - if I set the timeout higher - it just resets the next day.

I've also noticed that my battery is never fully charged in the mornings now - at 5:30am - it's about 80% full - rather than 100% as it's always been before - looking at the logs - it looks like predbat isn't sending the command to the battery to charge - even though it goes into charging mode.

Unfortunately my logs for today have rolled over but I'll grab them in the morning and add them to this issue.

Thanks
Leacho

leacho73, 2024-11-09:
Further to this - here are a few logs that appear to show that Predbat is not changing the mode on the battery - the battery should be charging but is currently discharging:

2024-11-09 01:35:23.781171: Warn: Historical day 7 has 390 minutes of gap in the data, filled from 21.53 kWh to make new average 29.53 kWh (percent 73%)
2024-11-09 01:35:23.882556: Warn: Inverter 0: Using inverter type SE - not all features are available
2024-11-09 01:35:26.061006: Info: record_status Charging target 72%-94%
2024-11-09 01:40:22.823580: Warn: Historical day 7 has 390 minutes of gap in the data, filled from 21.53 kWh to make new average 29.53 kWh (percent 73%)
2024-11-09 01:40:22.932495: Warn: Inverter 0: Using inverter type SE - not all features are available
2024-11-09 01:40:49.443437: Warn: Inverter 0: Using inverter type SE - not all features are available
Trigger callback for select select_option
2024-11-09 01:40:57.066827: Info: record_status Charging target 75%-92%
Trigger callback for switch turn_off
Trigger callback for switch turn_off
Trigger callback for switch turn_on
Trigger callback for switch turn_on
Trigger callback for switch turn_off
Trigger callback for switch turn_off
Trigger callback for switch turn_on
Trigger callback for switch turn_on
2024-11-09 01:45:23.528848: Warn: Historical day 7 has 390 minutes of gap in the data, filled from 21.53 kWh to make new average 29.53 kWh (percent 73%)
2024-11-09 01:45:23.615343: Warn: Inverter 0: Using inverter type SE - not all features are available
2024-11-09 01:45:25.714538: Info: record_status Charging target 78%-92%
2024-11-09 01:50:20.733308: Warn: Historical day 7 has 390 minutes of gap in the data, filled from 21.53 kWh to make new average 29.53 kWh (percent 73%)
2024-11-09 01:50:20.859549: Warn: Inverter 0: Using inverter type SE - not all features are available
Trigger callback for select select_option
Trigger callback for select select_option
2024-11-09 01:52:07.309064: Warn: Inverter 0: Using inverter type SE - not all features are available
Trigger callback for select select_option
Trigger callback for select select_option
2024-11-09 01:52:50.732649: Info: record_status Charging target 82%-100%
2024-11-09 01:55:24.148580: Warn: Historical day 7 has 390 minutes of gap in the data, filled from 21.53 kWh to make new average 29.53 kWh (percent 73%)
2024-11-09 01:55:24.238849: Warn: Inverter 0: Using inverter type SE - not all features are available
2024-11-09 01:55:26.683970: Info: record_status Charging target 82%-100%

[image]

richadlam, 2024-12-17:
I'm noticing the same issue, the plan shows that the battery should be set to charge, however no command is sent to the inverter to change mode at the time of the charge slot starting.
I am also using Solaredge inverter, with a solaredge battery
I also see the inverter resetting to defaults if the storage_command_mode is changed to anything other than 3600

springfall2008, 2025-02-09:
You have two options, either change your inverter settings to apply the command forever (rather than for 1 hour) or add 'repeat: True' to the charge start service in apps.yaml

closed 2025-03-08 by springfall2008