springfall2008/predbat_addon #28: Not executing plan

jrwm47, 2024-11-11:
Not sure if this is an “issue” or a misunderstanding. Predbat(8.5.11)’s plan for my GE AIO is almost never executed as shown. On a cloudy autumn day with negligible PV and a fairly steady and consistent day-to-day load (c.20kWh/day), the plan normally shows one or two large daily charge/discharge cycles, peaking to 100% SOC. But the predbat-controlled actual SOC rarely gets above 20%, even when the Agile import tariff goes -ve. In Calibration graphs, all forecasts seem highly volatile. h1 is invariably much lower than h8 and h12. Plan, graphs, yamls, settings, or parameters available if helpful.

gcoan, 2024-11-11:
You'll need to share more information on the problem as at the moment there's not enough to go on.

Firstly assume Predbat mode is set correctly to Charge & Discharge and it's not set to read-only?

Second, is your AIO responding to control requests via GivTCP OK?  If you look in the GivTCP log you should see commands when Predbat sets the charge rate, charge start time, etc.  These should all say 'written OK' to confirm that the inverter is receiving them.

Third, is Predbat working correctly.  Would need to see the Predbat plan, the logfile when the plan executes, and useful also to see the history of some of the key sensors in HA such as the AIO SoC and battery mode.

It's likely a simple config error on your part rather than Predbat doesn't work as it works fine for lots of us!

jrwm47, 2024-11-11:
Dear Geoffrey

Many thanks for your amazingly prompt reply. I agree that the issue is
almost certainly mine, not predbat's. Perhaps I should have posted the
issue on the community Facebook page but, as a grumpy old man, I won't use
Facebook on principle. I count myself a loyal predbat user and am
determined to get it running as well as possible. Now the evidence:

load:
    - large old stone house with oil c/h (Wiser controlled)
    - electrically-heated greenhouse and outbuildings (winter only)
    - Everhot 2k5W electric kitchen range (winter only c.600W loss to room)
    - myenergi eddi h/w PV diverter (summer only when c/h off)
    - EVs, bikes but no cars
    - retired occupants so daily load is fairly consistent unless we are
away

grid:
    - 1ph (G98 3k6 export  G99 applied for)
    - KAIFA MA120 smart meter (Octopus installed)

h/w & systems:
    - dedicated RPi4 running up-to-date HA
    - 3k6 Wp PV with Solis inverter
    - GE AIO *without *Gateway with Modbus CTs (nominally 13k5Wh but
manages 15kWh)

tariff:
    - Octopus Agile with fixed rate (15p/kWh) export

All CT power readings (GE, myenergi, KAIFI) agree within reasonable
margins-of-error

I attach a pdf with HA screenshots of the plan, graphs, entities, and files
you suggest. Predbat has been running and controlling my AIO for about 3
months. Prior to that I really struggled to get anything to work until GE
finally uploaded working firmware for the AIO *without* Gateway and
GivTCP V3 was released.

Today, for the first time in weeks, the sun is shining, the PV is
generating, and the peak SOC is higher than 20% but I guess it is the
Calibration graph with h12, h8, and h1 that puzzles me the most and what
happens when the sun disappears again.

Any suggestions you have would be most welcome.

Regards

Julian Morris
***@***.***

gcoan, 2024-11-11:
Hi Julian,  thanks for your message, hadn't appreciated that you have had Predbat working with the AIO for some months but your plan doesn't appear now to be executing correctly.

Looks like you replied to the github issue via email so all of your text has appeared in github OK (albeit not formatted very well) but the attachment has got stripped off.

Can you go to the issue directly on GitHub and upload the attachment please.

Re-reading your messages it seems as if there might be two different problems, one that the plan shows to charge to 100% but its not being executed and the SoC isn't getting above 20%,  and two that the calibration graph is volatile.  If the battery isn't charging when commanded to by Predbat then that could be the cause of the volatility, or it might of course not! My own calibration chart all the lines follow the same shape trend but the h8 and h12 are very 'sawtooth' up and down around the curve made by the actual and h1 lines.

jrwm47, 2024-11-11:
[predbat_screenshots_111124.pdf](https://github.com/user-attachments/files/17707447/predbat_screenshots_111124.pdf)
Geoffrey, I hope I've uploaded the screenshots correctly this time, Julian

jrwm47, 2024-12-14:
The issue I was attributing to _predbat_ appears to be solved. I discovered that _Givenergy parameter_ "**Battery Cutoff % Limit**" (only accessible through the portal under Remote Control) was set to 20%. I have no idea how and when this happened as I am fairly sure I had told the system to be set to Factory Defaults. Anyway, I have now set it to 100% and _predbat_ seems much happier.

closed 2024-12-14 by jrwm47

gcoan, 2024-12-14:
> The issue I was attributing to _predbat_ appears to be solved. I discovered that _Givenergy parameter_ "**Battery Cutoff % Limit**" (only accessible through the portal under Remote Control) was set [...]

This is interesting.  I was just adding this advice to the documentation, but checking both of my Gen 1 hybrid inverters, they are both set to 4%, the same as the battery reserve limit. And predbat works fine for me

jrwm47, 2024-12-15:
On the GE portal _Remote Control_ panel there are 2 parameters in the _Percentage Limits_ section:
    _Battery Reserve % Limit
    Battery Cutoff % Limit_
The first is, as you say, set to 4%, but it was the second that needed changing to 100%.

I think the idea is that you can cut off overnight charging if you sre sure there will be PV input the next day. Of course, predbat takes care of all that for you. Predbat and I are now both very happy.