springfall2008/predbat_addon #29: IOG car charging slots shorter than 30 minutes incorrectly processed

ianmk13, 2024-12-01:
IOG is scheduling fragmented charging sessions with fragments shorter than 10 minutes.  It appears that Predbat is designed around half-hour periods, so operation seems undefined. The attached files capture operation showing home battery charging outside of IOG-allocated period. (I've yet to see how this will appear on my bill).

[image]
[image]
[image]
[image]
[image]

springfall2008, 2025-02-09:
I don't see the actual issue, in that billing is in 30 minute segments so you get the whole half hour as low rate?

ianmk13, 2025-02-09:
It's a while since I raised this issue so details now hazy. Perhaps I was confused by Octopus allocating numerous con-contiguous mini-slots. Certainly my car didn't seem keen as it is reluctant to balance the cells at the end of a charging session in these conditions.
Happy for you to close the issue.

closed 2025-03-08 by springfall2008