springfall2008/predbat_addon #30: Predbat deciding I shouldn't charge for no discerible reason

SamKua, 2024-12-06:
I've been using Predbat for a few months now to manage charging my house battery, and its been working a treat. These last couple of weeks though its been behaving oddly randomly and decising not to charge, or to charge to very low levels, or sometimes deciding it will charge but then not let me use that charge and just sitting with the battery full pulling from the grid.

Currently, its decided I don't want to charge my battery tonight and instead I'll just run off the grid, in spite of having set a free electricity period tomorrow morning, and a lower rate overnight. 
[image]

I've not touched the settings from what had been working fine before, its set to control charge and on the latest update.

Currently I get round it by setting force charge slots when I notice its doing this (its not every day - last time it did this was Monday I think, and there have been another 2 or 3 days before then. 

Any ideas whats up?

SamKua, 2024-12-06:
And now, having added the force charge slots, its planning to hold the charge rather than letting me use it:
[image]

gcoan, 2024-12-06:
I see you have input_number.best_soc_max set to 9.5

This may be the cause of your problem.  I have mine set to the default 0.0, try changing it

SamKua, 2024-12-06:
Gave it a go, doesn't seem to have amde a difference. I think I set that over the summer in an effort to stop it charging the battery up more than necessary overnight then leading to us exporting when the solar did well. Obviously now, it needs to fully charge up every night since we are lucky to make 3kWh at the moment!

gcoan, 2024-12-06:
OK, worth a try.

Have a look in the Predbat web console at the configuration, and see what is set to a non-default value.  Something must be mis-configured for Predbat to not want to charge your battery up.

Have you looked in the logfile, are you seeing any warnings or errors?

jrwm47, 2024-12-08:
Your system seems to be behaving very much like mine (see issue #28). When the PV contribution is low, predbat doesn’t charge the battery even when the overnight tariff goes low or even negative.

What I noticed from the Calibration graphs is that the h12 and h8 plans both show full or significant battery charges, but when it finally gets to h0, the predicted charging just doesn’t happen. Is that how yours behaves?

Also like yours, in the summer, or even on the odd sunny day in winter, predbat seems to do what I would expect, charging from PV in the day and from grid at night if the tariff is low.

My best_soc_max is set to 0.

Julian

jrwm47, 2024-12-09:
[image]

This is what I mean. Does your calibration chart look like this?

Julian

SamKua, 2024-12-09:
That does sound similar my my issues. I've now reset the settings as much as possible to standard, I'm waiting to see if it behaves, with it being intermittent its hard to know for sure if anything has worked.

Our calibration chart has always looked crazy the further out we go. It does seem to suggest there are significant patches where it doesn't collect readings, which I wonder if that might be relevant?
[image]

jrwm47, 2024-12-14:
I am not sure if it is relevant to your problem but I have just closed Issue #28 with this Comment:

"The issue I was attributing to _predbat_ appears to be solved. I discovered that _Givenergy parameter_ "**Battery Cutoff % Limit**" (only accessible through the portal under Remote Control) was set to 20%. I have no idea how and when this happened as I am fairly sure I had told the system to be set to Factory Defaults. Anyway, I have now set it to 100% and _predbat_ seems much happier."

springfall2008, 2025-02-09:
Closing this issue as its stale, if it happens with the latest version please post a ticket in the Predbat respository as this one is just for the add-on.

closed 2025-02-09 by springfall2008