springfall2008/predbat_addon #33: Update notification loop going from 8.8.15 to 8.8.16

leacho73, 2024-12-27:
Predbat appears to be in an update notification loop whist trying to go from 8.8.15 to 8.8.16 - constant notifications and unable to apply the update automatically or via Home Assistant - the update button does nothing.

Log shows:

Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/predbat.py
Got data, writing to /config/predbat.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/config.py
Got data, writing to /config/config.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/prediction.py
Got data, writing to /config/prediction.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/gecloud.py
Got data, writing to /config/gecloud.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/utils.py
Got data, writing to /config/utils.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/inverter.py
Got data, writing to /config/inverter.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/ha.py
Got data, writing to /config/ha.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/download.py
Got data, writing to /config/download.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/unit_test.py
Got data, writing to /config/unit_test.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/web.py
Got data, writing to /config/web.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/predheat.py
Got data, writing to /config/predheat.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/futurerate.py
Got data, writing to /config/futurerate.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/octopus.py
Got data, writing to /config/octopus.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/solcast.py
Got data, writing to /config/solcast.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/execute.py
Got data, writing to /config/execute.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/plan.py
Got data, writing to /config/plan.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/fetch.py
Got data, writing to /config/fetch.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/output.py
Got data, writing to /config/output.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/userinterface.py
Got data, writing to /config/userinterface.py.v8.8.16
2024-12-27 12:27:20.114149: Info: Web socket stopping
2024-12-27 12:27:20.117911: Info: Web socket stopping
2024-12-27 12:27:20.118971: Info: Web socket stopping
Update complete
2024-12-27 12:27:38.583132: Warn: Historical day 7 has 530 minutes of gap in the data, filled from 16.29 kWh to make new average 25.78 kWh (percent 63%)
2024-12-27 12:27:38.619833: Warn: Historical day 14 has 730 minutes of gap in the data, filled from 10.84 kWh to make new average 21.98 kWh (percent 49%)
2024-12-27 12:27:38.789836: Warn: Inverter 0: Using inverter type SE - not all features are available
2024-12-27 12:27:54.699663: Warn: Inverter 0: Using inverter type SE - not all features are available
2024-12-27 12:27:54.755730: Info: Web socket stopping
2024-12-27 12:27:54.764125: Info: Web socket stopping
2024-12-27 12:27:54.896841: Info: record_status Charging target 14%-100%
File ./download.py was modified
Stopping Predbat due to file changes....
2024-12-27 12:27:54.912680: Stopping Predbat
Web interface stop called
Web interface stopped
Shutdown, sleeping 20 seconds before restarting
[12:28:17] INFO: Predbat init script running
Running Predbat inside Add-on
Your API key is: <REDACTED>
Bootstrap Predbat
Startup
Predbat files are installed correctly for version v8.8.15
**** Starting Standalone Predbat ****

gcoan, 2024-12-27:
If you turn predbat auto update off it should stop trying to apply the update whilst Trefor fixes what's wrong

Someone else has reported a problem with 8.8.16 upgrade https://github.com/springfall2008/batpred/issues/1796

leacho73, 2024-12-27:
Thanks for the reference @gcoan  - at least its not just me - unfortunately tuning off auto update hasn't stopped it trying - so will just wait for Trefor to issue a hotfix.

gcoan, 2024-12-27:
A hotfix is probably needed !

A bit more long winded, you can manually install a prior version of Predbat:

Ensure auto-update is turned off

Shutdown the Predbat addon

In the predbat directory /addons_config/6adb4f0d_predbat

Check there are none of the next version files hanging around, they'll be named eg predbat.py.v8.8.16 - delete these, leaving the .py files there

Download the release you want from github https://github.com/springfall2008/batpred/releases

Unzip the release

Copy all the .py files from the release ZIP over the top of the existing .py files in the predbat folder but don't over-write apps.yaml

Restart the predbat addon

symonchester, 2024-12-27:
Setting to main allowed me to turn off auto update

leacho73, 2024-12-27:
Spot on @gcoan  thank you, all updated and issue (appears) to be resolved.

closed 2025-02-09 by springfall2008