springfall2008/predbat_addon #34: Refresh config entities as config_refresh state is unknown

AberDino, 2025-01-04:
I had my GivEnergy Gateway and AIO installed in August last year and I am using it in combination with HA, GivTCP and Predbat. I'm on an import tariff which offers cheap energy at night (fixed) and I use Predbat in 'control charge' mode so that I can minimise export. This time of year, given the minimal solar output, this means the battery is charged to 100% overnight and is drawn down during the day.

Unfortunately, on two occasions recently, I've woken up to find the battery is at 4%, see [6adb4f0d_predbat_2025-01-04T10-14-38.022Z.log](https://github.com/user-attachments/files/18306459/6adb4f0d_predbat_2025-01-04T10-14-38.022Z.log) (trimmed). It appears the web socket is closed and Predbat is unable to reconnect after that, resulting in the log entry `Info: Refresh config entities as config_refresh state is unknown` repeated every 15 seconds. A restart of the Predbat add-on is required to resume normal service.

Ideally this issue would not occur of course, but as a minimum, would it be possible for Predbat to automatically restart when this log entry is recorded, or even send a notification so I can manually restart the add-on?

Thanks for all your efforts developing this excellent add-on!

gcoan, 2025-01-04:
There is a predbat error detection automation in the output data part of the documentation, this should restart predbat and givtcp if you get such issues

AberDino, 2025-01-04:
Thanks for the pointer, I shall have a look at that shortly.

AberDino, 2025-01-04:
Predbat error detection automation added [as per the documentation](https://springfall2008.github.io/batpred/output-data/#predbat-error-monitor), thanks again.

closed 2025-01-04 by AberDino