springfall2008/predbat_addon #35: Predbat giving "REST failed to setPauseSlot" message after recent update

RonS55, 2025-01-06:
I updated Predbat on the 4th January and I have since seen occasional Status values of "Inverter 0 REST failed to setPauseSlot".  I think this started with that update.
I have today updated to v8.10.0 and I am still getting this error.
I am running Predbat against a GivEnergy GIV-AC3.0 running software version D0.205-A0.205.

gcoan, 2025-01-06:
The AC3 doesn't fully support the battery pause mode feature, the fix is to comment out the start and end pause times from apps.yaml.

I've covered this in this draft change to the documentation https://github.com/gcoan/batpred/blob/main/docs/apps-yaml.md

RonS55, 2025-01-06:
Ah thanks.  I must have missed that in the upgrade notes when I updated.

RonS55, 2025-01-07:
Followed the instructions in the link above and this seems to have cured the issue.

closed 2025-01-07 by RonS55