springfall2008/predbat_addon #36: PV Forecast warnings in logs every 5 minutes

jon798, 2025-01-19:
Hi,

Running v1.2.2 addon (v8.11.1), on Raspberry Pi version of HA.

Also using GivTCP v2.4.9 (v3 won't install), Octopus and Solcast integrations.

The Octopus and Solcast integrations appear to be working correctly.

I am getting the follow warning logged every 5 minutes in the predbat log (the numbers changed at midnight) but are the same from the non in each log message:

Warn: PV Forecast data adds up to 17.06 kWh but total sensors add up to 2.48 kWh, this is unexpected and hence data maybe misleading (factor 6.9)

This is what's shown PV-related  in the predbat dashboard (Solcast API used count is none currently (it's 76:40am and dark), it reset from 10 around midnight):

PV Today | sensor.predbat_pv_today | 0.55 kWh | total0.55total100.26total900.97remaining0.55remaining100.26remaining900.97detailedForecast(large data)api_limitNoneapi_usedNone | total | 0.55 | total10 | 0.26 | total90 | 0.97 | remaining | 0.55 | remaining10 | 0.26 | remaining90 | 0.97 | detailedForecast | (large data) | api_limit | None | api_used | None

  | PV Forecast Now | sensor.predbat_pv_forecast_h0 | None W | now100.0now900.0 | now10 | 0.0 | now90 | 0.0

  | PV tomorrow | sensor.predbat_pv_tomorrow | 0.55 kWh | total0.55total100.12total901.75detailedForecast(large data) | total | 0.55 | total10 | 0.12 | total90 | 1.75 | detailedForecast | (large data)

  | PV day 2 | sensor.predbat_pv_d2 | 0.84 kWh | total0.84total100.16total902.77detailedForecast(large data) | total | 0.84 | total10 | 0.16 | total90 | 2.77 | detailedForecast | (large data)

  | PV day 3 | sensor.predbat_pv_d3 | 0.52 kWh | total0.52total100.17total901.24detailedForecast(large data) | total | 0.52 | total10 | 0.17 | total90 | 1.24 | detailedForecast | (large data)

Predicted PV Power | predbat.pv_power | None kW | results(large data)today0.0 | results | (large data) | today | 0.0

Predicted PV | predbat.pv_energy | 1.108 kWh | results(large data) | results | (large data)
results | (large data)

 Current PV kWh | predbat.pv_energy_h0 | None kWh

Predicted PV base 10% | predbat.base10_pv_energy | 0.384 kWh | results(large data)today0.26 | results | (large data) | today | 0.26

Predicted PV best 10% | predbat.best10_pv_energy | 0.384 kWh | results(large data)today0.26 | results | (large data) | today | 0.26

Predicted PV Power Best | predbat.pv_power_best | None kW | results(large data)today0.0 | results | (large data) | today | 0.0

Predicted PV best | predbat.best_pv_energy | 1.108 kWh | results(large data)today0.55 | results | (large data) | today | 0.55

The PV charts in Predbat look correct and the PV figures in the plan look correct.

I'm sure I've messed up some config somewhere - could anone point me in the right direction please?

gcoan, 2025-01-19:
I looked back in the github issues as I thought I recalled seeing this before, and I did, #1744 was the same warning message, but didn't get to the bottom of why it was occurring. 

I also had the same warning myself with #451 but this was a rounding error that @springfall2008 fixed in December 2023.

Can you attach a current logfile and debug file to aid with the investigation

jon798, 2025-01-19:
predbat_debug.yaml and logfile attached. Thanks.

[PV Log Warnings.zip](https://github.com/user-attachments/files/18469016/PV.Log.Warnings.zip)

springfall2008, 2025-02-09:
It's saying the numbers in the forecast sensor don't add up to the total of the sensor. Is your solcast actually updating correctly or perhaps its failed and the data is stale?

If the forecast is okay in Predbat then you can ignore it.

closed 2025-02-16 by jon798