springfall2008/predbat_addon #39: Web socket closed - Predbat dead

pacemaker82, 2025-02-05:
Got this in my error log today at 14.20, predbat never recovered, even after restarting HA. I restarted the add on some hours later and it finally recovered.

2025-02-05 14:24:48.327654: Warn: Web Socket closed, will try to reconnect in 5 seconds
2025-02-05 14:24:53.333494: Info: Start socket for url http://supervisor/core/api/websocket
2025-02-05 14:24:53.339906: Error: Web Socket exception in startup: Cannot connect to host supervisor:80 ssl:default [Connect call failed ('172.30.32.2', 80)]
2025-02-05 14:24:53.353208: Error: Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 980, in _wrap_create_connection
    return await self._loop.create_connection(*args, **kwargs)  # type: ignore[return-value]  # noqa
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1076, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1060, in create_connection
    sock = await self._connect_sock(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 969, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/lib/python3.10/asyncio/selector_events.py", line 501, in sock_connect
    return await fut
  File "/usr/lib/python3.10/asyncio/selector_events.py", line 541, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 111] Connect call failed ('172.30.32.2', 80)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/ha.py", line 184, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 805, in _ws_connect
    resp = await self.request(
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 562, in _request
    conn = await self._connector.connect(
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 540, in connect
    proto = await self._create_connection(req, traces, timeout)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 901, in _create_connection
    _, proto = await self._create_direct_connection(req, traces, timeout)
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 1209, in _create_direct_connection
    raise last_exc
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 1178, in _create_direct_connection
    transp, proto = await self._wrap_create_connection(
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/connector.py", line 988, in _wrap_create_connection
    raise client_error(req.connection_key, exc) from exc
aiohttp.client_exceptions.ClientConnectorError: Cannot connect to host supervisor:80 ssl:default [Connect call failed ('172.30.32.2', 80)]

2025-02-05 14:24:53.353848: Warn: Web Socket closed, will try to reconnect in 5 seconds
2025-02-05 14:24:58.359570: Info: Start socket for url http://supervisor/core/api/websocket
2025-02-05 14:25:17.865199: Error: Web Socket exception in startup: 400, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
2025-02-05 14:25:17.866269: Error: Traceback (most recent call last):
  File "/config/ha.py", line 184, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 821, in _ws_connect
    raise WSServerHandshakeError(
aiohttp.client_exceptions.WSServerHandshakeError: 400, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')

gcoan, 2025-02-05:
Most likely your HA access token had expired.  I have had this happen with similar fatal errors

The fix is to create a long lasting access token specifically for Predbat https://github.com/gcoan/batpred/blob/main/docs/apps-yaml.md#home-assistant-connection

pacemaker82, 2025-02-05:
OK interesting - so did restarting the add on recreate a token ?

gcoan, 2025-02-05:
> OK interesting - so did restarting the add on recreate a token ?

yes. it should be self managing but occasionally the automatic token times out.  I've made this more explicit in the (forthcoming) documentation update.

The error you get are connection errors from the websocket which aligns to this being the issue.  Nothing to loose by creating the token anyway

springfall2008, 2025-02-09:
Closing this as resolved

You can set the long lived token to avoid this issue if need be

closed 2025-02-09 by springfall2008