springfall2008/predbat_addon #47: iBoost Gas doesn’t switch on

PhillyGilly, 2025-03-22:
I have the settings below in my apps.yaml

‘’’
  # iBoost settings for control of Immerion heaters
  switch.predbat_iboost_enable: True
  switch.predbat_iboost_energy_subtract: True
  iboost_energy_today:
    'sensor.immersion1_2_energy_today'
  # Disable boost functions and enable gas price comparison
  switch.predbat_iboost_solar: False
  switch.predbat_iboost_charging: False
  switch.predbat_iboost_gas: True
  input_number.predbat_iboost_gas_scale: 1.0
  # Gas rates for comparison
  metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'
  #iboost_energy_scaling: 1.0
‘’’

Last night we had a few hours when the electricity price was lower than the gas price, but the binary sensor for iBoost active remained false.

gcoan, 2025-03-22:
> # Disable boost functions and enable gas price comparison
>   switch.predbat_iboost_solar: False
>   switch.predbat_iboost_charging: False
>   switch.predbat_iboost_gas: True
>   input_number.predbat_iboost_gas_scale: 1.0

You don't define switch and input_number values in apps.yaml, these have to be set directly in Home Assistant via integrations / entities / search for entity name

Or use a predbat dashboard as described in the documentation

https://springfall2008.github.io/batpred/output-data/#displaying-output-data

PhillyGilly, 2025-03-22:
Hi @gcoan I must have slept through the bit in the documentation that says to enable iBoost....
So I've got:
[image]

[image]

[image]

and apps.yaml becomes
```
  # iBoost settings for control of Immerion heaters
  #switch.predbat_iboost_enable: True
  #switch.predbat_iboost_energy_subtract: True
  iboost_energy_today:
    'sensor.immersion1_2_energy_today'
  # Disable boost functions and enable gas price comparison
  #switch.predbat_iboost_solar: False
  #switch.predbat_iboost_charging: False
  #switch.predbat_iboost_gas: True
  #input_number.predbat_iboost_gas_scale: 1.0
  # Gas rates for comparison
  metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'
  #iboost_energy_scaling: 1.0
```
Does that look OK now? Thanks

gcoan, 2025-03-22:
That looks a lot better

You might want to turn 'when iboost active prevent battery discharge' to stop your battery discharging to the immersion heater, but otherwise seems correct. 

Iboost when planned should appear in the predbat plan

PhillyGilly, 2025-03-22:
@gcoan thank you very much.

closed 2025-03-22 by PhillyGilly