springfall2008/predbat_addon #48: unable to update to 1.2.4

Metal-rock, 2025-04-03:
currently running 1.2.1 on a HA Green and cannot update, getting the same error everytime.
Error updating Predbat: The command '/bin/bash -o pipefail -c apt-get install -y python3-pip' returned a non-zero code: 100

gcoan, 2025-04-03:
Somebody else had a similar problem yesterday, and then all of a sudden it cured itself.  Might be worth retrying in a while (not satisfying I know)

Metal-rock, 2025-04-03:
i've been trying for weeks and get the same error everytime, is there a way to manually update ?

gcoan, 2025-04-03:
I don't know, beyond my docker knowledge, sorry

You can update predbat within the predbat addon using the select.predbat_version control,  updating the addon just updates the container that predbat runs in, so is less critical.

But can understand why you would want it upgraded to the latest version

Metal-rock, 2025-04-03:
I will live with it till the time comes for a fresh install and see what happens then

gcoan, 2025-04-03:
Here's the other issue that occurred yesterday, same error code: https://github.com/springfall2008/batpred/issues/2183

The OP had lots of attempts to get the install working, I would suggest you don't uninstall predbat as you may struggle to get it installed again like they did.

Suggestion was to try the install again and check the supervisor log for errors, and then for the OP it worked fine!

Metal-rock, 2025-04-03:
Just had a look in supervisor log and it looks like certain files aren't available

Get:109 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe arm64 python3-wheel all 0.37.1-2ubuntu0.22.04.1 [32.0 kB]
Ign:110 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe arm64 python3-pip all 22.0.2+dfsg-1ubuntu0.4
Err:20 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 binutils-common arm64 2.38-4ubuntu2.6
  404  Not Found [IP: 185.125.190.36 80]
Err:21 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libbinutils arm64 2.38-4ubuntu2.6
  404  Not Found [IP: 185.125.190.39 80]
Err:22 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libctf-nobfd0 arm64 2.38-4ubuntu2.6
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
 ---> Removed intermediate container dec7b322b77c

Metal-rock, 2025-04-15:
Fixed in the 1.27 update

closed 2025-04-15 by Metal-rock