springfall2008/predbat_addon #50: Battery detail not appearing in HA

PhillyGilly, 2025-04-15:
I have a Gen2 5kW hybrid with two 9.5kWh batteries.
Everything was running very well until early April but following re-install and upgrade, the problem has appeared.
Running GivTCP v3.1.6 under HA

[image]

The details of my batteries are appearing in MQTT e.g. battery 1

[image]

But not getting from MQTT into HA

[image]

Is there a problem with auto-discovery?

gcoan, 2025-04-15:
This is not a Predbat issue, likely it's with GivTCP and mqtt 

Are there any messages in the HA log about duplicate entity id?

Have seen this before if it is this:

Stop GivTCP 
With mqtt Explorer delete all the battery entities in the home assistant tree
Shutdown home assistant and all the add ons 
When it starts up again GivTCP should recreate the battery entities and fix the issue

PhillyGilly, 2025-04-15:
OOPS posted in wrong github. Idiot or what?

closed 2025-04-15 by PhillyGilly