springfall2008/predbat_addon #51: Changes needed for use in BE

GizmoBE, 2025-04-18:
In Belgium I need to use the EPEX prices. To use the EPEX add-on, I have to change the following files and rebuild Predbat to correctly read all values and display the correct output:

In octopus.py I change:
all_rates -> data
value_inc_vat -> price_per_kwh
valid_from -> start_time
valid_to -> end_time

In config.py and output.py I change:
currency-usd -> currency-eur
pounds -> euros
p/kWh -> c/kWh

gcoan, 2025-04-18:
You should have been able to set the currency rate Euros and cents in apps.yaml configuration to save editing the code. 

Sounds like the rates integration is a bit more bespoke, can you share the code after changing it to see how easy it is to add to core?

GizmoBE, 2025-04-19:
My currency seems to be correct in apps.yaml, but didn't display correctly on the plan page.

currency_symbols:
   - '€'
   - 'c' 

How can I share the code?

gcoan, 2025-04-19:
check the indentation of your currency settings in apps.yaml, if you don't get things indented correctly then they can be ignored by predbat.  You should be able to see the currency is set correctly from the predbat web console if you look at the apps.yaml tab.  If setup correctly it should be used in the plan

yes, attach the octopus.py code to this issue would be fine

cheers

GizmoBE, 2025-04-19:
Currency settings in apps.yaml seem correct. See the attached screenshot. I'll wait till the next update to check.
I also uploaded the octopus.py.

[image]

[octopus.zip](https://github.com/user-attachments/files/19820745/octopus.zip)

GizmoBE, 2025-04-20:
I just did the update and it looks like the plan tab is displaying correctly, but the dash tab isn't. Screenshots attached.

[image]

[image]

eddysteurs, 2025-06-01:
for the euro i have this in apps.yaml
  # Timezone to work in
  timezone: Europe/Brussels
  currency_symbols:
    - '€'
    - 'c'
  #
and I use the nordpool integration for the epex pricing
 # sensor.octopus_energy_electricity_xxxxxxxxxx_xxxxxxxxxxxxx_current_rate
  # sensor.octopus_energy_electricity_xxxxxxxxxx_xxxxxxxxxxxxx_export_current_rate

  metric_octopus_import: 'sensor.nordpool_kwh_be_eur_5_10_0'
  metric_octopus_export: 'sensor.nordpool_kwh_be_eur_5_10_0'

gcoan, 2025-06-01:
> I just did the update and it looks like the plan tab is displaying correctly, but the dash tab isn't. Screenshots attached.

Looking at your screenshot it looks like Predbat is correctly displaying the currency symbol on the plan and the unit of measurement for the money state entities.  This is probably the most important as these are the predbat outputs.

What I think you are reporting as not happening is the item and attribute names that contain a currency (e.g. p/kWh) do not use the currency symbol as defined in apps.yaml, they use UK default currencies.