springfall2008/predbat_addon #61: Grid use during peak import time when trying to keep batteries balanced

rrbiker72, 2025-09-17:
Setup :

1 x AC3 with 2 x 9.5kWh gen2s
1 x HY5.0 with 1 x 9.5kWh gen3

On an economy 10 meter which gives cheap rate 1am - 6am, 2pm - 5pm & 9pm - 11pm.

Prior to installing hybrid inverter the AC3 would cover the house load during peak import rate periods and recharge during the e10 periods. Obviously if the load exceeded 3kW the excess would be imported from the grid. With the 2 inverters they can cover a house load of up to 6.6kW which is perfect - just fired up the heat pump as it's getting a bit cooler now.

What I'm seeing though is predbat keeps throttling the discharge rate on the HY5, I think to keep the batteries balanced, but this means the larger load is being supplied from the grid when the discharge rate changes to 0W. This means I'm then importing at the very expensive rate while the battery sits there waiting for the other battery to discharge to the same level. I get that this is because the HY5 only has one battery and therefore its charge % will reduce at twice the rate of the AC3's 2 batteries but I wanted to know if there's a setting in predbat that can overcome this issue?

I've changed the input_number.predbat_balance_inverters_threshold_discharge to 5% to see if this helps. 

Are there any other settings I can tweak to limit the import at peak times under a heavy load?

Thanks

rrbiker72, 2025-09-17:
Yes I am. I used the balance inverters automation when I first installed the second inverter and that was ok but I'd been using predbat all year and liked the solar prediction so decided to get that up and running. 

It's been fine and only now the loads are heavier it's having a problem. I've got RTC on so the writes are all on safe registers and I keep checking the other writes which are static so it seems ok. I know what you mean about the number of writes though...it's in the 1000s!

I think givenergy are working on a HEMS that is similar to their EMS which I will invest in when it's available. Not sure if it will be able to manage both AC coupled and hybrid inverters together....if not I'll have to invest in another hybrid inverter to replace the AC3 and use it in AC coupled mode. 

Where can I find the other issues you've raised?

gcoan, 2025-09-17:
> I think givenergy are working on a HEMS that is similar to their EMS which I will invest in when it's available. Not sure if it will be able to manage both AC coupled and hybrid inverters [...]

I believe they are, but the EMS for AC took so long to arrive that I wouldn't anticipate it arriving any imminent time.
I also expect they are firstly going to focus on EMS for hybrid only, and then mixed mode with hybrid and AC coupled will be a later iteration - if ever, depends on market size and whether its worth developing and testing, etc.

If it ever does come along I am not at all convinced I would buy it TBH.  The AC HEMS is ~ £600 so by the time its fitted with installation effort it'd be £1000 and whether I would see that benefit, I am not sure,

I have two hybrid inverters and have pretty much learned to live with how they operate together.   I don't bother with wanting the battery levels to be matched, I just let them do their own thing.   When the inverters are both charging or both force discharging there is no issue, they behave fine.  In Eco (Demand) mode they work fine during the day when its sunny and in fact when one battery is full the other inverter will use excess AC to charge its own battery if it needs to rather than letting it be exported.  When there's house load to meet they share it, usually inverter 1 takes more load, but I see them both discharging eg when the kettle or oven comes on.

The issues come when there's no sun and a house load reduces (e.g. kettle turns off).  One inverter can see the excess discharge/export from the second and starts to charge from it, the second sees that there is still demand and continues discharging and the two inverters thus cross charge which isn't efficient.
For me the cross charge is fairly gentle and its always from inverter 1 to 2, so I have decided to live with it now.  I have got an automation that stops the cross charge and I used it for a while, this was before real time registers, and maybe if I was really concerned I would resume using it.

What stops the cross charge very well is setting the discharge rate on inverter 1 to zero or pausing the battery charging. Of course 5 minutes later predbat undoes this.  Hence my FR for predbat to be able to set charge rate to zero when there is no solar and no planned charging.  I could write an automation to instruct predbat to do this but would prefer it be part of core predbat